bitfocus/companion-module-panasonic-ptz #1: Feature request: add Power On/Off and Install Position command

andreaimpact, 2019-10-03:
It would be useful to have the following commands.
- Power ON 
- Power  OFF
- Switch to Desktop  Install Position 
- Switch to Hanging  Install Position

JeffreyDavidsz, 2019-11-24:
Power on/off is in the latest build now, could you test it?

andreaimpact, 2019-11-25:
Hi Jeffrey,

I downloaded the latest win64 version companion-2.0.0-b6fc4d5-1883-win64 but I don’t see any ON/OFF action? Am I missing something?

Thank you

Andrea

JeffreyDavidsz, 2019-11-27:
It seems the build server is full, so no updates from 18-11 are stored there. When cleaned download the latest, then it's there

andreaimpact, 2019-11-27:
Yes, after writing you I noticed that the latest buld on repository is dated 18.11 and therefore that the module was not updated yet.

Thank you anyway

Andrea

andreaimpact, 2019-12-12:
I confirm that Power On/Off is working. 
Still missing 
Switch to Desktop Install Position
Switch to Hanging Install Position

Thank you!

McHauge, 2020-11-29:
Next module update will include the option to change the mode from desktop to hanging position, still need to test it though, hope to do so in the next weekend.

McHauge, 2020-12-21:
This is now a part of the Beta builds, will close this for now, feel free to open it again if need's be

closed 2020-12-21 by McHauge

airbenich, 2020-12-21:
Great! Thank you @McHauge Andreas!