bitfocus/companion-module-panasonic-ptz #4: AW-HE40 Record

videochris, 2020-01-14:
Is it possible to start and stop the record function on SD card for the HE40?

McHauge, 2021-02-17:
@videochris Sorry, but I can't find the command for it if you find let me know and I'll add it to the list of actions.

daniep01, 2021-02-17:
Here are the commands. If you have access to the camera then you can find them using the network tool in chrome/safari web inspector.

/cgi-bin/sdctrl?save=start
/cgi-bin/sdctrl?save=end

McHauge, 2021-02-17:
Well I don't have access to it, i only have the aw-ue150. But I'll try to add it

McHauge, 2021-02-17:
also, did you find it mentioned in any documentation?

McHauge, 2021-02-17:
I think I found it: https://eww.pass.panasonic.co.jp/pro-av/support/content/guide/DEF/HE40_UE70_CGI/Supplement_for_Web_Control-UE70E.pdf

McHauge, 2021-02-17:
@daniep01 I have now added an "SD Card Recording" action, but no feedback, as I don't see a great way to do it without adding a third way for the module to get updates. This would (from what I can see) require polling the camera every 1 sec or so. But if we are lucky it will report in the already established subscription-based system that is in place, though this is not documented in it, so...

I'll upload a Windows build to the link below in 10-15 min, and if you could perform a test with it and enable debugging (in the instance config page) that would be awesome. then look at the "info" and "warn" logfile when you start and stop a recording from Companion as well as from the web interface. if there is any data coming back in the log file while you do this, then please copy or screenshot it to me, then I might be able to add feedback based on it.

[OneDrive](https://1drv.ms/f/s!AtUYbExJseNtlK56u-n71V_UcLDe7g)

McHauge, 2021-02-18:
@daniep01 The core just got updated, so you should now be able to get this in the newest beta. But please keep my comment above in mind and let me know how it goes.

daniep01, 2021-02-18:
Thanks, I can test at the weekend but I think it was @videochris who was originally asking for this feature, I just let you know the url because I had discovered it already.

daniep01, 2021-02-20:
Hi @McHauge,

I have just tested and I see the following error in the log

```
20. 21:36:50 instance(HE40-1): Error executing action: self.sendWeb is not a function
20. 21:34:03 instance(HE40-1): Error executing action: self.sendWeb is not a function

```

McHauge, 2021-02-21:
I'll give it a look

McHauge, 2021-02-21:
@daniep01 Found the error, I forgot to push the change from index.js where I declare the function :) but it should be fixed now, and be in the beta after the next poll

daniep01, 2021-02-21:
I'll look out for the next build

daniep01, 2021-02-21:
@McHauge 
Recording Start/Stop looks good to me on HE40. I tried enabling the debug log but didn't see anything interesting, not sure if this is expected.

McHauge, 2021-02-21:
@daniep01 great to hear, no I did not expect any, but if there were that would have been awesome, because then feedback would have been an easy addition,

McHauge, 2021-02-25:
I'll close this, as this is now fixed in the core, and should work as expected in all new beta releases

Feel free to open it up again if something comes up

closed 2021-02-25 by McHauge