bitfocus/companion-module-panasonic-ptz #6: Up and Right Controls Don't Work After Setting P/T Speed

Stu232, 2020-03-31:
This is definitely a problem within companion because the up and right controls will work if the speed is not set in the module, however once it is set for the companion module then they will no longer work within companion. The down and left controls continue to work. If I create a new module then they work again. I would just do this but then they are set to the wrong speed and wouldn't work for me. Also the entire time my physical ptr controller works fine.

philipdroberts, 2020-11-10:
Any status update on this? I can confirm the bug is still present with the following software versions: 

I've also noted that the up and right controls will start working again after making + or - to PT speed.

HE130
Firmware version       
    CPU Software       
        Interface V02.20
        Camera Main V02.20
        Servo V01.01
        Network V02.38
    EEPROM     
        Interface V01.00
    FPGA       
        Lens V01.02
        AVIO V01.04
        Com V01.02

Companion v.2.1.0 (2.1.0-47823d0-2493)

Thanks

Philip

McHauge, 2020-11-29:
I think this should be solved in the next version of this module, but I need to do some final testing.

McHauge, 2020-12-21:
Seems to be semi-fixed, the first time it needs to presses (when changing speed) currently but the ptz will move no matter what will do some more testing to see it I can get the last part to work correctly, but it's better that before

philipdroberts, 2021-01-27:
I tested build ending in #2713 or #2714 yesterday and it seems to work as expected for the features I tested.

After a "set PT speed" action I can use all 4 directional controls and they work as expected.

McHauge, 2021-01-27:
The changes I'm working on is not included in the betas, only in my builds, so not to sure what you testet 😅

McHauge, 2021-01-27:
Ups sorry thought is was another comments, but anyway I'm working on a big update with feedbacks and variables for this, you can see the changes in the development branch on the module

McHauge, 2021-01-27:
And if you want to test them, please take a look at this facebook post https://m.facebook.com/groups/companion/permalink/2786070918277907/?sfnsn=mo

philipdroberts, 2021-01-27:
No worries, about getting threads mixed up.

Having feedback based on the "Preset number query command" in companion would be great to know which preset was recalled last. 

As evidenced by how long it took me to try your change from back in December I don't know that I have time to do much testing your builds. Sorry.

McHauge, 2021-01-27:
No worry's, but feel free if you find the time for it

McHauge, 2021-02-18:
@philipdroberts Just wanted to say, that the core has been updated with the newest changes and this should now be fixed in the latest beta builds. but the feedback you are talking about has not been added, but it's a lot simpler to do now, with all the back end part for feedbacks and variables built-in.

McHauge, 2021-02-25:
I'll close this, as this should now be fixed in the core, and should work as expected in all new beta releases

Feel free to open it up again if something comes up

closed 2021-02-25 by McHauge