bitfocus/companion-module-panasonic-ptz #7: [Module Request] Panasonic PTZ - One Touch Auto Focus

dvdenz, 2020-05-28:
Hello,
Would it be possible to enable the One Touch Auto Focus function of the Panasonic PTZ Cameras? I am specificially using the AW-HE40 camera.
The workaround I am using right now is to send HTTP GET commands - one to turn auto focus on ( http://10.0.0.55/cgi-bin/aw_ptz?cmd=%23D11&res=1 ) and one to turn it off again ( cmd=%23D10&res=1 ) once it's had time to auto focus approx 850ms later.

I have tried sending the HTTP GET command to trigger the OTAF function directly ( http://192.168.0.10/cgi-bin/aw_cam?cmd=OSE:69:1&res=1 ) but the camera just jumps back to auto focus. Unsure if implementing it in the module would result in proper behaviour? Thanks!

Edit: I think the behavior changes depending if the camera is already in AF or MF mode

McHauge, 2020-11-29:
Added support for one-time autofocus, but still, need's to be tested, and, I'm near to our PTZ until next week, but if it works out, all get it included into the core.

McHauge, 2020-12-21:
This is now a part of the Beta builds, will close this for now

closed 2020-12-21 by McHauge