bitfocus/companion-module-panasonic-ptz #8: Not working

duckjerkdan, 2020-06-15:
I am trying to use the module with bitfocus 2.0 and it is not working. When I add the IP of my camera nothing happens and the instance status says NULL. I am using an AW-HE40. 

Do I have to do anything on the camera side? I can access the web interface of the camera using the IP address so I know the IP is correct. Any thoughts?

McHauge, 2020-12-21:
I got a fix for this, will get included in v1.0.8 of the module.
A fix so far is to disable and re-enable the module after setting the IP. Then it should work as expected

McHauge, 2021-02-18:
@duckjerkdan Hi, the core just got updated with the latest version of the module, with a major overhaul. the AW-HE40 should now have better support, just follow the instance config page while setting up the camera. The beta builds from now on includes this update. Therefore I'll close this for now.

closed 2021-02-18 by McHauge