bitfocus/companion-module-panasonic-ptz #11: Panasonic PTZ add Preset recall Mode

TreborKutscher, 2020-11-11:
Please add a Dropdown for the Recallmode to a specific PTZ Preset on panasonic cameras.
Or make it an individual command to set the recall mode.

There are Mode A / B / C wich allow to not recall everything stored in the preset. Presets include gain  white ballance and collor information wich are allo stored in a preset memory. 
In some cases you want only recall position zoom and focus of the camera while doing live color correction on the fly with the rcp panel. 

it can bea found on the manual of the remote controller rcp 120 from panasonic;
https://pro-av.panasonic.net/manual/pdf/AW-RP120G_Operations(VQT4T93A-1(E))_E.pdf Page 21

This my also be an dynamic variable wich should be able to read out. An innstance feedback might also be handy.

I would like to use it as follows:
One Presets or multiple ones should host the presets for white ballance gain and all the colorsettings. Which should be recallable via companion. For this specific preset/s i want them to recall all values. I will recall them before show start to reload color settings or if a scenic change requires an other color set. 

Then i have a lot of presets wich  only exist to call camera positions and zoom values. but need to leave the color settings untouched.  This is neccassry during an event over the day which is daylight affected. The color changes will drift over the day and the operator can change it like an normal with the rcp on the fly. while he can recall the stored positions without recalling the now uncorret color  

@PerRoine

josephdadams, 2020-11-11:
I have transferred this to the Panasonic PTZ repository where someone can follow up with you.

McHauge, 2021-06-23:
@TreborKutscher Sorry about the wait and no response, This is now added to the module and will be available in the next release, as well as new v2.2.0 betas in a week or so.

although this is not added as a dropdown on the recall preset action itself, its added as a separate action and feedback, if you want to you can always stack them on the same button :) there are also premade presets available in the "Preset Recall" folder under presets.

closed 2021-06-23 by McHauge