bitfocus/companion-module-panasonic-ptz #13: Support for Panasonic AW-HE38

bpatel928, 2021-01-26:
Hey there,

Currently there does not seem to be any support for the Panasonic HE38 PTZ. I do not think that there is much of a difference between it and the HE40, but can a instance be made for that camera?

McHauge, 2021-01-26:
There is a major update in the works, there is even an open alfa for testing support on all panasonic PTC's
This can be found on both facebook and slack. You can see the changes if you check the "development" branch of the module

McHauge, 2021-01-26:
https://m.facebook.com/groups/companion/permalink/2786070918277907/?sfnsn=mo

McHauge, 2021-02-18:
@bpatel928 and @kraves-oz the core just got updated with the latest version of the module, with a major overhaul. the AW-HE38 should now b supported in the beta builds from now on. therefore I'll close this for now.

closed 2021-02-18 by McHauge