bitfocus/companion-module-panasonic-ptz #14: WHITE BALANCE

studiovideo54, 2021-02-08:
Hello, why preset white balance is not present?
Many thanks

Dory

McHauge, 2021-02-09:
@studiovideo54 Cause they haven't been added yet... it's possible to add them in the new development branch, but in the current main branch would be hard, as there are 3-5 types of ways Panasonic does white balance on its PTZ's.

jbilyj, 2021-05-21:
It would be great to see at least variable colour temperature control added. I know this would extremely useful for the AW-HE130's that we run.

McHauge, 2021-05-21:
Yep, it's definitely on the list, just haven't had any camera i could take home to test on for a while

cpuguru07, 2022-10-19:
Also needing White Balance/color controls.

studiovideo54, 2022-10-19:
> Reply to: bitfocus/companion-module-panasonic-ptz ***@***.***>

Yes for my panasonic awhe130

 

Doriano Costantino

Studiovideo54

Cell. +39 338.8747472

***@***.***

premultiply, 2026-08-27:
This module is deprecated and is no longer actively maintained.

Implemented in its successor, https://github.com/bitfocus/companion-module-panasonic-cameras.