bitfocus/companion-module-panasonic-ptz #15: LUMIX BGH1 Support request

PhotoJoseph, 2021-02-10:
I have no idea how open or available the resources needed would be, but the new [BGH1](https://pro-av.panasonic.net/en/cinema_camera_varicam_eva/products/bgh1/index.html) (box camera) can be controlled completely over ethernet using the LUMIX Tether app… so theoretically, it could be controlled in Companion!

McHauge, 2021-02-17:
@PhotoJoseph If you can find documentation of the protocol then we can look at If it's possible. But with the need for an app, I don't think it would be an easy implementation.

chberge, 2021-06-11:
There is some kind of SDK/API for the BGH1 available here: https://av.jpn.support.panasonic.com/support/global/cs/soft/tool/sdk.html

McHauge, 2021-06-11:
@chberge Just looked at it, it seems it's a .dll and that would basically block it from being used in companion as it's only for windows on the site, and Companion is windows/mac/Linux. Even if possible, it would share nothing from this module so should be made into its own module.

PhotoJoseph, 2021-06-11:
oh shoot sorry @McHauge I totally missed your original reply. Interesting that there's documentation but on windows only. Let me see what I can find out now that we know there's at least a windows implementation.

hamuchen, 2022-03-31:
As @McHauge stated, this feature will probably need its own module and I made a request nearly at the same time as @PhotoJoseph here:
https://github.com/bitfocus/companion-module-requests/issues/402

I think we could work it our over there, if you want to help, it will be greatly appreciated. I think out setups (from what I can tell from your videos) are very similar. 

Cheers, 
hamu

closed 2023-07-12 by josephdadams

Manouchehri, 2023-07-12:
Support was added?

josephdadams, 2023-07-12:
Duplicate request

also 
https://github.com/bitfocus/companion-module-panasonic-lumix