bitfocus/companion-module-panasonic-ptz #19: (Bug) Panasonic AW-UE150s Won't Go to Preset 1

BrianMSimons, 2021-02-25:
I setup a really long automated series of camera movements using the Preset Recall action and having vMix mix the shots together.  Everything works perfectly until at the very end of the list I Recalled each camera to go to Preset 1, that way I can just start the sequence again seamlessly like a loop.  All of the other Preset numbers work, but the log for when you try to preset #1 gives this:  

25. 16:55:36 instance(PTZ-3): Send CMD: Rnull

It's an easy workaround for me, I'm going to avoid preset 1.  Just wanted to share the bug.

McHauge, 2021-02-25:
Did you use a preset or add the action your self, also did you try and select another preset number and then go back to the nr 1 preset in drop-down on the same action

BrianMSimons, 2021-02-25:
All of the presets I created using the RP150 controller.  Then in the photos attached, you'll see just the very top of my long sequence in the image named Companion Sequence Example.  I have redundant Drive Time actions before every movement, so that I can easily change the speed of this sequence later.  It runs through Three cameras with 22 Preset positions each.  And again, works perfectly except when asking it to go back to Preset Position 1.

In my other screengrab of the the simple button - I made a button that only had three actions.  Tell each camera to go to Preset Position 1.  None of the cameras respond.  I changed it to Preset 3, or any other Preset - and the work every time. 

[image]

[image]

McHauge, 2021-02-25:
Yes but if you take one of those actions, and with out adding a new one or removing any, thy and thanked the drop-down menu to preset 1 i feel like it will work then, it's just the "default" value is not working, and I'll give that a look

BrianMSimons, 2021-02-25:
Hmm....I see what you're saying now.  I may have just left it as is since it showed Preset 1.  I'm going to check now....  and you are correct.  If I actually grab another preset from the drop down, then select Preset 1 - it works.  Good call.

McHauge, 2021-02-25:
Thx for trying it, it does not solve the problem altogether, but it makes its usable, I'll try and see if I can get it patched in a day or two, should be a simple fix. Just missed a few things under testing after the many changes 😊

BrianMSimons, 2021-02-25:
I love how quickly you are on top of these things.  I am not tech savvy
enough to know what's going on under the hood, but give me a few weeks
farting around and I'll try to automate nearly everything and if I find any
quirks I'll keep on posting.  Cheers

McHauge, 2021-02-25:
Thanks, I just pushed a fix, also found 6 other actions that also had missing defaults, so those should also be fixed in the next pull into the core.

McHauge, 2021-02-28:
@BrianMSimons  just tested it, and it works fine here, please check your actions and verify that they are set correctly they should look like this (also added a picture of my log when testing): 

[image]
[image]
[image]

McHauge, 2021-02-28:
lol wrong thread sorry

BrianMSimons, 2021-02-28:
No prob.  Have a goodnight.

McHauge, 2021-03-26:
@BrianMSimons did you test the fix pushed 29 days ago?

McHauge, 2021-04-23:
@BrianMSimons did you get to test the build yet? I'll close this issue, and if you feel like it, feel free to open it up again iff need be.

closed 2021-04-23 by McHauge

BrianMSimons, 2021-04-23:
Hi Andreas, everything appears to be working perfectly with the PTZ
presets.  I've been continually grabbing newer beta releases and using
them, and I haven't had any issues with critical functions. One of the more
recent builds had countdown variables for Black Magic Hyperdecks, those do
not work for me with my Hyperdeck Minis.  If I put them in the name of a
button it just shows -:--:--, even when a clip is playing.

Sorry I haven't responded sooner.  All of my work is at a local university,
and with all the virtual convocations coming up I'm moving non-stop.  I
hope to test more and make even better automation once school's out and I'm
not working just to keep my head above water.

Thanks,
Brian