bitfocus/companion-module-panasonic-ptz #20: Autofocus off issue on Panasonic PTZ

DonSlotta, 2021-02-27:
Hi there ,
I´m controlling Panasonic AW-UE70 and AW-HE130 PTZ cameras via Companion.
I´m able to turn the Autofocus on via the _**Focus Mode: Auto Focus**_.  But the command _**Focus Mode: Manual Focus**_ doesn´t turn it off. Is this a bug?

McHauge, 2021-02-27:
@DonSlotta first of all, what version of companion?
if you are on the beta, please try and enable debug to log and restart companion
then press the action and see what it reports in the log

DonSlotta, 2021-02-27:
I´m using version v2.2.2(2.1.2-f4bcaf3-2679)

McHauge, 2021-02-27:
please ty the Beta, the module has been rewritten with feedbacks and all

to download it click the "builds" button under the main download button: [link](https://bitfocus.io/companion/download/builds/)

DonSlotta, 2021-02-27:
Thanks for for the link. I will try it. Do I need a specific build or just the latest? 
Sorry for these newbie questions.

McHauge, 2021-02-27:
just the latest

another thing if you are on windows go to this folder %AppData%\Roaming\companion
and make a backup off all it's content. this is all your settings and buttons in companion, if something breaks, you can just download the release version again, and copy your backup back into this folder and you will be back to what you have now

DonSlotta, 2021-02-27:
I tried version v2.2.0 (2.2.0-7cb0beb-2932).
Now both commands **Lens - Focus Mode: Auto Focus** and **Lens - Focus Mode: Manual Focus** turn the Autofocus mode on. None of them turn it off.

McHauge, 2021-02-27:
would you mind enabling "debug to log" and restart companion
then press the action and see what it reports in the log
then post a screenshot of it

DonSlotta, 2021-02-28:
I sent both commands for Auto Focus and Manual Focus. Both semm to send the same value. 

[image]

McHauge, 2021-02-28:
Thx I'll give it a look

McHauge, 2021-02-28:
@BrianMSimons  just tested it, and it works fine here, please check your actions and verify that they are set correctly they should look like this (also added a picture of my log when testing): 

[image]
[image]
[image]

McHauge, 2021-02-28:
@DonSlotta   just tested it, and it works fine here, please check your actions and verify that they are set correctly they should look like this (also added a picture of my log when testing): 

[image]
[image]
[image]

DonSlotta, 2021-02-28:
I´ve double checked the action on my Auto Focus Off Button. It is the same action as in your test setup. See attached screenshot
Your action triggers the Send CMD: D10 command. Mine still the D11 like the Auto Focus On command. That´s weird.
[image]

McHauge, 2021-02-28:
Well, my bet then would be to make a backup of the setup, then try removing the instance an create new buttons, else try a completly clean setup and se if that helps

DonSlotta, 2021-02-28:
I deleted the instances, created them new and made new buttons. Now it works. Except that the manual focus command triggers the Auto Focus on and the auto focus command turns it off. That's still weird but I totally can live with that. Thanks a lot for your support. It helped me a lot. Cheers.

commit 82e262fc (2021-04-07): V1.0.14 - Bugfix and Updated Help.md

Fixes #20
closed 2021-04-07 by McHauge