bitfocus/companion-module-panasonic-ptz #21: Using Companion and AW-RP50 together

daniep01, 2021-02-28:
Hi,

Maybe this isn't a real issue because it is all working as designed but I note that when Companion and the RP50 are both used in the same setup the camera indicator lights on the RP50 flash continuously. The manual says this means that the camera is also connected to another controller as well as the RP50 which is true but the continuous flashing is annoying to me. I'm just wondering if the module could behave differently to prevent this from happening? Is it triggered by the TCP polling maybe? I would be happy to do some experiments but I'm also happy to accept it's all working as it should.

JeffreyDavidsz, 2021-02-28:
It's as panasonic has designed it. That way you can see something else is also controlling your controller. If you only use it for tally, set that in the ptz module and leave the RP50 alone.
I'll close this issue as this is a panasonic design

closed 2021-02-28 by JeffreyDavidsz

daniep01, 2021-02-28:
Maybe it is as Panasonic intended but if I use the companion http module to talk to the cameras it doesn't happen, so I will need to revert to that method.