bitfocus/companion-module-panasonic-ptz #24: Please add support for Panasonic CX350/CX4000

dieterjanssens, 2021-03-26:
If possible to integrate:

https://eww.pass.panasonic.co.jp/pro-av/support/content/guide/DEF/CX350_Command_for_PTZ_Control_Protocol.pdf

McHauge, 2021-03-26:
@dieterjanssens Quick question does it do anything if you select the type other or auto?

(you will need companion 2.2.0 beta for that)

also could you please go to the pages below and save the content of that page in a text file and send it to me for reference.
http://YourCamerasIP/live/camdata.html
http://YourCamerasIP/cgi-bin/get_basic
http://YourCamerasIP/cgi-bin/getinfo?FILE=1

this would help a lot, in order to understand, if feedbacks could be adopted

dieterjanssens, 2021-03-27:
Thanks to Todd Kaczorowski for the txt file
[cx350 info.txt](https://github.com/bitfocus/companion-module-panasonic-ptz/files/6216046/cx350.info.txt)

My cx350 is in back-order so for the moment without cx350 tot test

McHauge, 2021-04-07:
@dieterjanssens Did it give any response on http://YourCamerasIP/live/camdata.html
Or is that just a dead link on the device? as that is the link used for feedbacks currently.

also for this to work you will need to have NDI|HX enabled, as per the documentation.

dieterjanssens, 2021-06-12:
Finnally I have a cx350 on hand with NDI|HX enabled. Unfortunatly http://YourCamerasIP/live/camdata.html gives a 404 error, so dead link.

McHauge, 2021-06-24:
@dieterjanssens From what I can see, this should be possible but the initial loading of variables from the camera might be a bit of a pain, as the command above gave you a 404 error. and that is how I currently fetch the current state when a connection is made.

dieterjanssens, 2021-12-16:
Update, i just found this document.
I hope this can help us to integrate the cx350 in Companion.
https://eww.pass.panasonic.co.jp/pro-av/support/content/guide/DEF/CX350_CX4000_Command_for_PTZ_Control_Protocol.pdf

dieterjanssens, 2021-12-20:
> Update, i just found this document. I hope this can help us to integrate the cx350 in Companion. https://eww.pass.panasonic.co.jp/pro-av/support/content/guide/DEF/CX350_CX4000_Command_for_PTZ_Control_Protocol.pdf

> @dieterjanssens From what I can see, this should be possible but the initial loading of variables from the camera might be a bit of a pain, as the command above gave you a 404 error. and that is how I currently fetch the current state when a connection is made.

R3aL1Ty13, 2022-07-10:
Hello, any news here ? :-)

Rik83, 2022-08-03:
hey guys, your module is now working with panasonic CX350. zoom ok iris ok but some bugs for the other functionnaliky. Can you update ?

dieterjanssens, 2022-08-05:
> hey guys, your module is now working with panasonic CX350. zoom ok iris ok but some bugs for the other functionnaliky. Can you update ?

It's working! on our cx350!
Iris up / down is working also zooming in / out ND filter and gain is also working!
Only auto iris button acts as manual iris
and 
Manula iris button acts as auto iris
Pedestal is not working
Further investigation will get more details.
Would it be possible to display/read-out the iris settings / ND filter settings / Gain settings?

PR #72 (2023-12-15): Add CX350/4000 series support
Add basic support for

- AG-CX350
- AG-CX200
- AJ-UPX360
- AJ-CX4000
- AJ-UPX900

According to https://eww.pass.panasonic.co.jp/pro-av/support/content/guide/DEF/CX350_CX4000_Command_for_PTZ_Control_Protocol.pdf

Please note that the firmware of these camera models does not (yet?) support event update notification.
Therefore, there are no variables or feedback for these models as no status updates are transmitted from the camera back to companion.

Fixes #24
closed 2023-07-12 by josephdadams