bitfocus/companion-module-panasonic-ptz #25: Request: instance feedback for AW-HE42

vetle-munin, 2021-04-23:
Is it possible to get instance feedback to see what preset is active and to read out the values for gain, iris, pedestal, etc.?

McHauge, 2021-04-23:
@vetle-munin Sure, but I'll have to test it out on one of our cameras, and haven't had one close for some time now. I'm not sure if the last preset thing is possible, based on if a preset is loaded from another controller, but if that gets passed through then surely that can be added.

before adding values that are different for all cameras, I would want some basic white balance control implemented, but that will take some time. The thing is a lot of the values change between cameras so it would be like 5-15 different lists of values or setting that would need to be set up. But it's on the list :)

vetle-munin, 2021-04-23:
@McHauge that is super, thanks alot for looking into it! :)

closed 2023-07-12 by josephdadams