bitfocus/companion-module-panasonic-ptz #27: Dont register uncaughtException handler

Julusian, 2021-06-28:
https://github.com/bitfocus/companion-module-panasonic-ptz/blob/80fbe3e7c64f2dad045e634fd38e1292939b4e08/src/index.js#L295

I don't think this module should be doing this. This is a global event handler, meaning if any module in companion doesnt handle an exception this module will be asked to handle it.
This will change the default error behaviour of companion where it shows the dialog then continues to exiting (https://github.com/bitfocus/companion-module-panasonic-ptz/blob/80fbe3e7c64f2dad045e634fd38e1292939b4e08/src/index.js#L350), and if any other module/instance was the one to trigger the 'EADDRINUSE', then every panasonic-ptz instance will assume it was for them and do things

Also, the event handler isnt being removed, meaning that each time `init_tcp()` is called, another handler will be added and a reference to the class will be leaked.

McHauge, 2021-06-28:
Already have it removed in a local comit from today, I'll push it in a moment

McHauge, 2021-06-28:
Do to the problem mentioned on slack, I have completely redone the init_tcp() but the problem still persist

Julusian, 2021-06-28:
Also instead of the port offset stuff you are doing, have you considered letting node choose a port for you? If you supply 0 as the port number, it will choose a free one.
You can then get the port number it chose with `socket.address().port`

McHauge, 2021-06-28:
> Also instead of the port offset stuff you are doing, have you considered letting node choose a port for you? If you supply 0 as the port number, it will choose a free one.
> You can then get the port number it chose with `socket.address().port`

that's interesting, and definitely, what i wanted in the first place, I did this as I didn't know how to get a "random" port and receive it again. I'll look into changing it over to that instead :)

Julusian, 2021-06-28:
> that's interesting, and definitely, what i wanted in the first place, I did this as I didn't know how to get a "random" port and receive it again.

you're not alone there, Ive seen a similar pattern in a bunch of modules. This is probably something that should be documented better on the wiki, although it is already there in the nodejs documentation

McHauge, 2021-06-28:
done, just pushed a change over to random port from Node, instead of an offset

McHauge, 2021-06-28:
also has better handling of subscriptions, so less chance of leaving a subscription open on a camera

McHauge, 2021-07-02:
I'll close this, as it now fixed and included in the core

closed 2021-07-02 by McHauge