bitfocus/companion-module-panasonic-ptz #34: [feature request] Buton combination support for moving in diagonal directions

07pepa, 2022-03-03:
We are using streamdeck to control panasonic PTZ camera.
On one of screens we implemented inverse T arrows (like on keyboard) to move it manualy and it works.

However there is no way to program it in way to use two buton combination to go in any of diagonal direction

i think folowing "buton" combination should be allowed and interpeted as such no matter of order how they are pressed
- ⬆+➡ = ↗
- ⬆+⬅ = ↖
- ⬇+➡ = ↘
- ⬇+⬅ = ↙

Maybe it is limitation of companion itself but i think it should not be.
Please Allow this
Josef

josephdadams, 2022-03-03:
Can't you use the Preset button definitions for this module to move diagonally? We have two Panasonic PTZs and this is what we do... works great. Unless I'm misunderstanding

07pepa, 2022-03-03:
If you mean predefined diagonals butons technicaly I can use it.
However they will not fit in our 5 x 3 stream deck since we have preset recals speed selects and zoom control and it leaves two button free and that is not enought for all 4 diagonal combination.

If Preset buttons are meant to work in combination being pressed they don't (and  form quick code glance hereit does not look like it).

I also verified that conpanion is detecting both buttons being pressed at same time.  And from my experience i know
this can be implemented.
Just hold current status of buttons in some relevant variable and do folowing actions under described logic
- if one buton is presed just go on with that direction
- if valid combination is  pressed figure out what diagonal it corespond to and start sending for that diagonal
- on last relased buton stop moving (for fluid movement when transioning betwen combinations)
- if invalid combination is pressed keep moving in last know good direction or stop

Same logic i used for controling other ptz camera with keyboard combinatons (but code is old and ugly when i was learning coding)  ( https://github.com/07pepa/DAHUA_HTML_CONTROLER ) 
And i did not use JavaScript for a while so i PR would take ages. 

Thanks for undestanding and sorry for long explanations and hints how to potentialy do it i will undestand if this will be just closed.

Josef

07pepa, 2022-03-10:
This one of way to implement it (i did not test it because as i said i did not research much your setup and to to build it for myleft and test in in companion) 

but i think you can get idea from code how to do it  and i only included changed files you can diff them easyly 
but it is not properly formated

for above mentioned reason i am not raising PR since code is poor quality but 

https://drive.google.com/file/d/1HVkkQCtp0rQYacwbcBQDhf3H5ZC5Lmkd/view?usp=sharing

dnmeid, 2026-08-15:
This can be solved by your configuration, but would be hard to implement on module level. Closing as not planned.

closed 2026-08-15 by dnmeid