bitfocus/companion-module-panasonic-ptz #36: Request: Feedback for Active/Last Recalled Preset

cpuguru07, 2022-03-15:
Would it be possible to add feedback for the current active (or last recalled) preset?

cpuguru07, 2022-10-27:
The current/last recalled preset appears to be available on line 19 of camdata.html and formatted "s#" (s0 = home; s1 = preset 1; s94 = preset 94).

premultiply, 2024-01-08:
Should be possible. I'll have a look for this as I was missing that on my last multi-PTZ production.

Poncho42, 2024-06-21:
Was just curious if this ever went anywhere. Would love to see this get implemented!

premultiply, 2024-06-21:
Ready to use: https://github.com/bitfocus/companion-module-panasonic-cameras/releases/tag/v1.0.0

Just waiting to be included in the beta.

cpuguru07, 2024-06-21:
@premultiply Is there a reason for this being a separate module instead of added to this existing module? I have extensive commands in this module, and I'm not looking forward to reprogramming all of my buttons to use the new module.

premultiply, 2024-06-21:
You can read about the reasons here: https://github.com/bitfocus/companion-module-panasonic-ptz/pull/81

Poncho42, 2024-06-21:
> Ready to use: https://github.com/bitfocus/companion-module-panasonic-cameras/releases/tag/v1.0.0
> 
> Just waiting to be included in the beta.

Oh awesome! Any idea when it will hit a stable version of bit focus? I clearly won't be able to use it this weekend if it requires a complete rebuild of my bit focus instance. But it would be great to implement before the next show. 

This weekend is 12 UE150 and would be great to see at a glance what preset was recalled. 

Normally it's just 6. 🙃

premultiply, 2026-08-27:
This module is deprecated and is no longer actively maintained.

Implemented in its successor, https://github.com/bitfocus/companion-module-panasonic-cameras.