bitfocus/companion-module-panasonic-ptz #40: AW-HE40 user authentication prevents module from working

tsdtheatres, 2022-07-05:
I have user authentication enabled on my 3 PTZ cameras. 

Normally when sending IP commands you just add the username/password to the IP address. user:password@192.168.1.101 for example.   

I can add the module and it connects "OK" with the companion app but the commands don't work.  If I disable user auth on the cameras, the module works as intended.  I'd like to leave authentication enabled on the cameras. Is it possible to use this module with an AW-HE40 with user authentication enabled?

mikethiessen21, 2023-03-04:
I would also like this module to work with user authentication enabled.

SNRSE, 2023-07-17:
+1 on this request currently running into an issue with user auth and the UE100

cidermillaudio, 2023-07-17:
+1 on this issues

tsdtheatres, 2023-10-07:
Discovered today on the newest 3.1.0 update that I can connect to the cameras WITH their authentication enabled! In the "camera IP" field in the module settings, enter the user/password in front of the IP like this: username:password@192.168.1.100  

Not sure how long that's been working but I am very glad to finally be able to control my PTZ's via companion!

premultiply, 2026-08-27:
This module is deprecated and is no longer actively maintained.

Implemented in its successor, https://github.com/bitfocus/companion-module-panasonic-cameras.