bitfocus/companion-module-panasonic-ptz #44: How to create tally properly in companion 3.0

DVEntertainment, 2023-01-09:
Hi, since the companion 3.0 update, this line appears in the connection settings tab:
"Tally On (Basic)
Support for Tall On is no longer possible. Instead you can set this up as a trigger, and get additional control"

I was really happy with the way the tally worked in companion 2.4.1, i used it all the time in combination with various atems. But i need companion 3.0 and i'm unsure how to build a tally trigger the right way, because however i try it, it hangs too often. (tally wont come on or doesn't go off, because my trigger doesnt get triggered).

 What is the right way to build a trigger for tally with an atem mixer?

For clarity i use a 2m/e constellation HD in combination with 5 AW-HE42's.

Thanx in advance!

Restheap, 2023-05-10:
You need a http module configured. 
I think one is enough as long as you include the full url in each command. 
Make a set of triggers from feedback "Tally: Program" and  "Tally: Preview".

Set red tally on: http://x.x.x.x//cgi-bin/aw_cam?cmd=TLR:1&res=1

And one with the same feedback inverted for Red tally off:
/cgi-bin/aw_cam?cmd=TLR:0&res=1

For green/preview the same apply:
/cgi-bin/aw_cam?cmd=TLG:1&res=1
/cgi-bin/aw_cam?cmd=TLG:0&res=1

Restheap, 2023-05-10:
Make a set of triggers from ATEM feedback "Tally: Program" 
Action: Panasonic-PTZ: System - Tally On

And one with the same feedback inverted for  tally off: