bitfocus/companion-module-panasonic-ptz #52: Tally Feedback ON not working

SOSF-Regie, 2023-03-23:
The tally feedback when tally is on is not working. Always show off.

Bitfocus v3.0.0

cyprient, 2023-07-12:
Can you please give your camera model?

UKAtlas, 2023-08-25:
@cyprient 
Just to add, I have just run into this issue.
AW-UE150 (v2.97)
Companion v3.0.0

Have tested and done a work around using generic http requests which work.
http://192.168.0.10/cgi-bin/aw_cam?cmd=TLR:1&res=1
http://192.168.0.10/cgi-bin/aw_cam?cmd=TLR:0&res=1
For Red Tally on and off both work.

Trigger (does not work):
On condition becoming true
atem: Tally: Program
Panasonic-PTZ: System - Red Tally On

Trigger (does work):
atem: Tally: Program
httpLocal: GET (above URLS)

cyprient, 2023-08-25:
@UKAtlas Can you please give me the version of the module you use? And is the module configured to auto detect your camera model, or are you specifying the model manually?

UKAtlas, 2023-08-26:
@cyprient - I'm away from my pc today, but we tried auto and selecting the camera, neither worked. Module is the default that comes with the companion installer.

cyprient, 2023-08-26:
@UKAtlas As different module versions were built in Companion v3, can you give me the module version when you can?
I'll investigate the issue.

cyprient, 2023-10-17:
> Also experiencing this with AW-UE150 running Firmware V03.08.
> 
> 
> 
> Companion version 3.1.1+6229-stable-4593dd7f and Panasonic PTZ module v.2.0.0.
> 
> 
> 
> No errors or log entries generated when trying to change the tally

@nriebesehl-pega The version of the Panasonic PTZ module built in Companion 3.1 is not 2.0.0 but 2.2.0. Can you please recheck the version you have?

nriebesehl-pega, 2023-10-17:
> //www.pega.com/>
>
>
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
>
> CAUTION: This email was sent from an EXTERNAL source. Think before clicking links or opening attachments.

My mistake, it is v2.2.0. Sorry about that!

Nick Riebesehl | AV Production Lead, Executive Briefing Center | Pegasystems Inc.

One Main Street, Cambridge, MA 02142

Office: (857)

cyprient, 2023-10-17:
> My mistake, it is v2.2.0. Sorry about that!
>
>
>
>
>
> Nick Riebesehl | AV Production Lead, Executive Briefing Center | Pegasystems Inc.
>
>
>
> One Main Street, Cambridge, MA 02142
>
>
>
> Office: (857) 856.3948 | [...]

Do you have any error in the logs when you try to set the tally on or off?

nriebesehl-pega, 2023-10-18:
No, we do not see any errors or log entries generated when sending the command to turn the tally off or on.

cyprient, 2023-11-04:
> No, we do not see any errors or log entries generated when sending the command to turn the tally off or on.

@nriebesehl-pega Can you please join a capture of the module logs after having performed the following actions:
- Red Tally On
- Red Tally Off
- Green Tally On
- Green Tally Off

UKAtlas, 2023-11-06:
I should be able to test on Wednesday.

PR #71 (2023-12-15): Fix Red/Green Tally ON action
Fixes #52
closed 2023-12-15 by bryce-seifert