bitfocus/companion-module-panasonic-ptz #53: Feedback Power State OFF not working

SOSF-Regie, 2023-03-23:
The feedback for Power Sate Off not working. Always ON.

cyprient, 2023-08-02:
Do you still have this issue? If yes, can you please give your camera model and the Companion version you run?
I have not been able to reproduce the problem on my configuration.

plansar, 2023-11-07:
I have the same issue - power state variable is always ON, does not change to OFF.
Companion v3.1.2, same with v3.0.1
Panasonic AW-HE42, firmware Ver.2.1

cyprient, 2023-11-07:
> I have the same issue - power state variable is always ON, does not change to OFF.
> 
> Companion v3.1.2, same with v3.0.1
> 
> Panasonic AW-HE42, firmware Ver.2.1
> 
> 

Can you please send the logs you have when you power on/off the cameras?

plansar, 2023-11-08:
> > I have the same issue - power state variable is always ON, does not change to OFF.
> > Companion v3.1.2, same with v3.0.1
> > Panasonic AW-HE42, firmware Ver.2.1
> 
> Can you please send the logs you have when you power on/off the cameras?

It looks that restarting the connection to camera, i.e. stop and start connection helped. Now it works fine.

premultiply, 2026-08-27:
This module is deprecated and is no longer actively maintained.

Fixed in its successor, https://github.com/bitfocus/companion-module-panasonic-cameras.