bitfocus/companion-module-panasonic-ptz #56: How to send custom commands ?

Camille92, 2023-05-09:
Hello there and thank you for your work!

I'm planning to acquire the new Streamdeck+ and would like to use it as a Remote Operation Panel.

I'm interested in controlling camera settings such as temperature, matrix, gain, etc. with just a touch of a button.

I understand that it might be challenging to include all commands as presets in Companion, but would it be possible to implement a "custom command" button that sends the code I find on this website to my cameras?

https://eww.pass.panasonic.co.jp/pro-av/support/content/guide/EN/top.html

I know that I can send the command using generic HTTP requests, but I believe this method would be more cumbersome in my opinion.

premultiply, 2024-01-08:
What commands are missing exactly for your operation?

Camille92, 2024-01-09:
For PTZ feature :
full support of the AWUE50
Auto tracking on/off

I wanted to convert a Streamdeck + into a kind of AK-HRP1000 and use the rotary buttons to change reb green blue gain.

premultiply, 2024-01-13:
I'll have a look to add camera support, red/blue gain+pedestal and rotary support.

premultiply, 2026-08-27:
This module is deprecated and is no longer actively maintained.

Implemented in its successor, https://github.com/bitfocus/companion-module-panasonic-cameras.