bitfocus/companion-module-panasonic-ptz #64: [BUG] Cameras loosing connection / stopping connection

linecheck, 2023-08-05:
Hi everybody,

First of all, thanks for the awesome module and the great work all of you are doing. Unfortunately I have troubles with our Panasonic-PTZs and Companion. After a while the cameras loose the connection to Companion. The NDI-stream still works, the connection between the cameras and the controller and between vmix stays, only companion has troubles. I added a picture of the error and the part of the log, but I can´t make anything out of it. Has anybody an idea? Or where to look? I asked this in the slack channel and I was told it is probably a bug, since the log file says something about "cannot read property of "series"".

Thanks Bastian

P.S.: I´m sorry, but I couldn´t find out how to do the proper formatting for bug reports, any help for next time is appreciated ;)

[image]
[image]

cyprient, 2023-08-05:
Hello @linecheck,
Could you please indicate which version of the module you are using? When did the bug first occur?

linecheck, 2023-08-07:
Hello @cyprient and everybody else,

sorry for the late reply. We are on v2.0.3. We installed it with Companion on June 12th. But the error already existed prior to that, when we were on the beta version of Companion 3.0.

I must admit that we are using three AW-UE160 which might not be supported (although everything works when the connection exists), but camera 4-6 are AW-UE80 which are also affected by the error. By the way, the error doesnt happen with all cameras at once, but one by one (not even in order).

I´ve seen the module is at version 2.1.0 so I will make an update on thursday, maybe the bug is already gone, but still maybe somebody can check...

Thanks

Bastian

linecheck, 2023-08-08:
Hello @cyprient,

that´s odd, we started our setup with only the three UE-80 and everything worked fine for almost half a year, so I never thought of it much. The problem started when we added the UE-160. Is there a reason, why these cameras are not supported? Shouldn´t the api be the same?

We have AUTO TCP checked and have "auto detect" enabled...

Thanks

Bastian

cyprient, 2023-08-08:
@linecheck Can you set the model to UE-150 (for your UE-160 cameras connections) and tell me if you see any improvement?

LotsOfRaptors, 2023-10-18:
@linecheck, did you manage to find a resolution to this issue? I am having the same problem with an AW-UE50

cyprient, 2023-10-19:
> @linecheck, did you manage to find a resolution to this issue? I am having the same problem with an AW-UE50

Can you please tell me which version of Companion and Panasonic PTZ module you are using?

LotsOfRaptors, 2023-10-30:
> > @linecheck, did you manage to find a resolution to this issue? I am having the same problem with an AW-UE50
> 
> Can you please indicate which version of Companion and Panasonic PTZ module you are using?

I am using Panasonic: PTZ Camera v2.2.0 on Companion [3.1.1 (3.1.1+6229-stable-4593dd7f)]

cyprient, 2023-10-31:
> > > @linecheck, did you manage to find a resolution to this issue? I am having the same problem with an AW-UE50
> 
> > 
> 
> > Can you please indicate which version of Companion and Panasonic PTZ module you are using?
> 
> 
> 
> I am using Panasonic: PTZ Camera v2.2.0 on Companion [3.1.1 (3.1.1+6229-stable-4593dd7f)]

Thank you for the information. Can you also please share the logs you have when you encounter the issue?

Hennyr, 2024-11-02:
Since i Use version beyond 3.2.2 i have often loose connection. I get red dots and the connection tab display loss of connection.
Reconnect thru the connectiontab works sometime longer or shorter time. 
The version 3.4.1 is the connection more worse. Go back to 3.2.2

dbongartz2001, 2025-01-10:
@linecheck we had the same problem with our Panasonic AW-UE50, because they aren't supported. 
For us it works just to take the AW-UE63 as model.

premultiply, 2026-08-27:
This module is deprecated and is no longer actively maintained.

Fixed in its successor, https://github.com/bitfocus/companion-module-panasonic-cameras.