bitfocus/companion-module-panasonic-ptz #74: Tally not triggering since 3.0.0

ready1take1, 2024-01-03:
I use the trigger function to turn on/off all the camera tallies and ever since the July 15th version of 3.0.0 it has not worked. I haven't changed anything but as soon as I downgrade to the June version of 3.0.0 it works. 

It looks like the info in the connections tab has been updated. Under system instead of "Tally State" it's changed to "Tally On/Off". The support for the green tally has been added but it is not working for me.

Can anyone else confirm this? Thanks

premultiply, 2024-01-08:
Tally works fine but you need to go the way and setup two triggers for each tally on/off on each camera.

Vesa-Stremia, 2024-01-10:
Using ATEM Constellation 8K + Panasonic AW-UE150 cameras
Companion version 3.2.0+6637-beta

I've set triggers as follow:

Cam1 Tally ON:
Event: On variable change ($atem:pgm1_input)
Condition: atem: Tally: Program: Input = Cam1
Action: Cam1: system - Red Tally On

Cam1 Tally Off:
Event: On Variable change ($atem:pgm1_input)
Condition: atem: Tally: Program: Input = Cam1 INVERTED
Action: Cam1: System - Red Tally Off

For some reason with this, I get Cam1 Tally On when Cam1 is in PVW.
If I add more cameras they get mixed so that e.g. When Cam1 is in PGM, Cam4 Tally is On.
And yes, I've triple checked that the camera IP:s are correct.

I tried to just invert Conditions in every trigger, and for one camera at time that seems to work, but yet again multiple cameras messes the system totally.

premultiply, 2024-01-10:
Tally is something different than the Input on program bus!

There a special tally feedbacks from the mixers tally logic for that usecase.

Vesa-Stremia, 2024-01-10:
But isn't the Input on PGM what defines if Tally should light up or not? So that when the trigger finds cam1 on the pgm input, it sends the signal to cam1 to turn the tally light on?

Since this seems to be such a mystery, could you tell us how to set the trigger properly? I tried to search the web but there are no instructions anywhere.

ready1take1, 2024-01-10:
Yes that is how it is suppose to work, and how I have been using it up until these last couple versions of Companion have been released. But for me I can't even fire a command through companion that involves tallies. I can do everything else though but not that. For now I am just going to stick with 3.0.0 since it is working there. 

It used to be so much easier when the tallies would turn on/off without needing triggers.

premultiply, 2024-01-10:
It could be much more simple if there will be a direct mapping of a event / variable to a boolen action but until that will be available you have to define to triggers for each camera like this:
[image]

Vesa-Stremia, 2024-01-11:
Thank you for the image, that helped. Now I have all cameras and tally's working 100%

cyprient, 2024-01-11:
The issue #44 contains information on how to set tally triggers in Companion 3.x
https://github.com/bitfocus/companion-module-panasonic-ptz/issues/44

closed 2024-01-11 by cyprient