bitfocus/companion-module-panasonic-ptz #93: Loosing connection in module PTZ Optics - visca

Hennyr, 2024-11-04:
Since i Use version beyond 3.2.2 i have often loose connection. I get red dots and the connection tab display loss of connection.
Reconnect thru the connectiontab works sometime longer or shorter time.
The version 3.4.3 is the connection more worse. Go back to 3.2.2

I have done some extra tests and in version 3.4.3 the joystick function (thru the PTZ Optics app & bluetooth) did not work any more or the commands did not responds good (moving to up = shooting 180 degrees up)

In version 3.3.0 the connection was sometimes disconnected or Companion gave red dots, but the camera was still responding for 30 minutes.

jswalden, 2024-11-07:
Did you mean to file this against the panasonic-ptz module, or against the ptzoptics-visca module?

Hennyr, 2024-11-07:
I have deleted my message in Panasonic - vista and moved toe PTZ Optics 
Vista. Had mad a mistake.

Sorry

---
Met vriendelijke groet,

Henny Ribbink

-------- Oorspronkelijke bericht --------

  		ONDERWERP:
  		Re: [bitfocus/companion-module-panasonic-ptz] Loosing connection in 
module PTZ Optics - visca (Issue #93)

  		DATUM:
  		07-11-2024 04:16

  		AFZENDER:
  		Jeff Walden ***@***.***>

  		ONTVANGER:
  		bitfocus/companion-module-panasonic-ptz 
***@***.***>

Did you mean to file this against the panasonic-ptz module, or against 
the ptzoptics-visca module?